Qrome/printer-monitor #146: Working but no temp Info from Octoprint

Vossberger, 2021-10-16:
Installation on Wemos and http-Server running. Screen is working probably.
Display shows "online" but I have "0°F and NA" if I start a print ther is no change. 
API key + IP is correct. Web-Server shows "Status : Operational"

When I press the button "Test Connection and API JSON Response" I get the following message :

{"job":{"estimatedPrintTime":null,"filament":{"length":null,"volume":null},"file":{"date":null,"name":null,"origin":null,"path":null,"size":null},"lastPrintTime":null,"user":null},"progress":{"completion":null,"filepos":null,"printTime":null,"printTimeLeft":null,"printTimeOrigin":null},"state":"Operational"}

What is wrong with my installation ?
Printer-Monitor V 3.0 / Octoprint v 1.7.0 @ Phython 3.6

Qrome, 2021-10-16:
Make sure you have a valid and activated weather API key... you must activate it after creating it.   Looks like you are not getting the weather API call response.   You can also verify this by viewing the Serial Monitor set to baud 115200 -- you can see the logging and responses there for the API calls.

Vossberger, 2021-10-17:
Thank you, Weather + starting a print and "test connection" resolved my "issue".

closed 2021-10-17 by Vossberger