Qrome/printer-monitor #151: Percentage Difference Between Printer Monitor and Octoprint Web Interface

SillyHats, 2021-11-11:
Since the Print Time Genius plugin adjusts the percentage complete from the default (which is much more accurate in my experience), the percent shown on the printer monitor does not match the octoprint web interface. This is a request to implement a similar solution to that linked below in an octoprint plugin.

https://github.com/pwnbus/OctoPrint-Progress-Title/issues/2

Qrome, 2021-11-11:
The only issue I have with this is it appear that the Octoprint server would require the Time Genius Plugin -- I do not plan on implementing solutions that require plugins to be installed.  Ife I am missing something, please let me know.

SillyHats, 2021-11-11:
Thanks for taking a look at this. It seems like the plugin where this issue was previously noticed and addressed uses an if/else statement to work with or without print time genius.

https://github.com/pwnbus/OctoPrint-Progress-Title/pull/5/files#diff-d974064e041ed343df7b2140d7ee9aad715070f99150401c87a4d7f095d15cb9

Qrome, 2021-11-11:
Yeah, I was looking at that.  You could easily update your own code to do the same thing with the response data.