Qrome/printer-monitor #154: Screen gets messed up after a while

Zeleni-Zub, 2021-12-11:
Hello,

I followed the instructions and it works for a while. Not sure why but screen gets messed up after a while and from what I could notice it gets messed up after a pause to get the time/weather data. Problem is that it randomly starts doing it, sometimes it works ok for a couple of hours and sometimes it does it much sooner. On the image I uploaded it seems like screen is upside down but while it worked it is in proper orientation.
Only thing I did to the code is enter the needed data (API keys and such) in Settings.h prior to upload but I noticed the same behavior if I upload the code with no changes. 
SSD1306 on Wemos D1 mini. Hardware seems to be ok as it works normally for a while but with no experience in programming I cannot say it is the code. Not sure what is happening here :( 

[image]
r

Qrome, 2021-12-11:
Never ever seen this.  Please check all the wires are cleanly connected and not shorting.  I have a feeling it is an issue with the connection to the OLED.

Zeleni-Zub, 2021-12-11:
I've checked and they look ok (with multimeter). Screen end is connected with jumper cables and wires on the Wemos end are soldered. Also if there is a short it wouldn't work ok for a while then go berserk? When I plug it in it works just fine.
I found that it doesn't get messed up during the pause for time and weather data collection but happens randomly.
Anyways I borrowed another Wemos from a friend since all of mine are not available and re-did everything but using the same screen.
Looks ok so far, it's been on for ~30 mins. If it stays that way then there's definitely something wrong with the first Wemos.
I'll post an update in a day or two after it has been turned on the whole time.

Zeleni-Zub, 2021-12-11:
Looks like I found a culprit, the screen! I used another Wemos but with the same screen and it started acting again. Only reason why I tried with the same screen was because it used to work fine on my Raspberry Pi so I didn't suspect it can be the cause. Using the white SSD1306 now and it has been turned on w/o any issues for over 5 hours now. I don't have another blue/yellow OLED so I can't tell is that the issue but it shouldn't make a difference. I will be using a blue one in the future.
Thank you for your time and thank you for such a cool and practical project!

closed 2021-12-11 by Zeleni-Zub

Qrome, 2021-12-11:
Awesome!   Glad you got it all worked out.