Qrome/printer-monitor #155: Restarting Issue

SohaibSiddique, 2022-01-14:
Hello, 
I have just set up the Wemos D1 mini and it's restarting after uploading the program, 
it almost restarts after every 10 sec.

SohaibSiddique, 2022-01-14:
I have decoded the exception and here is the result. do not know how to fix that.

```
Decoding stack results
0x4021fde5: String::invalidate() at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h line 295
0x402040b8: OpenWeatherMapClient::updateWeather() at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/WString.h line 79
0x4020dda8: WiFiClient::connect(char const*, unsigned short) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src/WiFiClient.cpp line 131
0x40204511: TimeClient::updateTime() at /Users/sohaib/Downloads/printer-monitor-master/printermonitor/TimeClient.cpp line 37
0x40223abc: uart_write(uart_t*, char const*, size_t) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/uart.cpp line 546
0x4021d594: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x4021d5a0: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x4021d594: HardwareSerial::write(unsigned char const*, unsigned int) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/HardwareSerial.h line 193
0x4021e868: Print::write(char const*) at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.h line 59
0x4021eadc: Print::println() at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/Print.h line 57
0x4020564e: getUpdateTime() at /Users/sohaib/Downloads/printer-monitor-master/printermonitor/printermonitor.ino line 420
0x40204118: TimeClient::getCurrentEpoch() at /Users/sohaib/Downloads/printer-monitor-master/printermonitor/TimeClient.cpp line 147
0x402084bc: std::_Function_handler  ::setup(esp8266webserver::ESP8266WebServerTemplate *, String const&, String const&, String const&)::{lambda()#3}>::_M_invoke(std::_Any_data const&) at /Users/sohaib/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/xtensa-lx106-elf/include/c++/10.3.0/bits/invoke.h line 103
0x40206140: getMinutesFromLastRefresh() at /Users/sohaib/Downloads/printer-monitor-master/printermonitor/printermonitor.ino line 1281
0x4020ced8: loop() at /Users/sohaib/Downloads/printer-monitor-master/printermonitor/printermonitor.ino line 378
0x402214f4: loop_wrapper() at /Users/sohaib/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266/core_esp8266_main.cpp line 201

```

Qrome, 2022-01-14:
Try compiling it with the ESP8266 Core that is described in the Readme file.

Also, from looking at your first post, it looks like it is failing making a connection to the weather API.

SohaibSiddique, 2022-01-15:
Thank you for your reply actually the weather status because I was not set the weather API, 
as per the Readme file, I have installed the given version of esp board and when I flash it for the first time it's working but after switching on and off it's not working its keeps restarting.
I had run some other scrip on my D1 mini and those were fine, 
following is the screenshot of my board setting

[image]

SohaibSiddique, 2022-01-17:
@Qrome my recent finding is code restart when a time update request is generated,
if I comment out the following code it's working fine,

```
if (!timeClient.connect(ntpServerName, httpPort)) {
    Serial.println("connection failed");
    return;
  }
```

here is the full code block for updating the time

```
void TimeClient::updateTime() {
  WiFiClient timeClient;
  
  if (!timeClient.connect(ntpServerName, httpPort)) {
    Serial.println("connection failed");
    return;
  }

  // This will send the request to the server
  timeClient.print(String("GET / HTTP/1.1\r\n") +
               String("Host: www.google.com\r\n") +
               String("Connection: close\r\n\r\n"));
  int repeatCounter = 0;
  while(!timeClient.available() && repeatCounter < 10) {
    delay(1000);
    Serial.println(".");
    repeatCounter++;
  }

  String line;

  int size = 0;
  timeClient.setNoDelay(false);
  while(timeClient.connected()) {
    while((size = timeClient.available()) > 0) {
      line = timeClient.readStringUntil('\n');
      line.toUpperCase();
      // example:
      // date: Thu, 19 Nov 2015 20:25:40 GMT
      if (line.startsWith("DATE: ")) {
        Serial.println(line.substring(23, 25) + ":" + line.substring(26, 28) + ":" +line.substring(29, 31));
        int parsedHours = line.substring(23, 25).toInt();
        int parsedMinutes = line.substring(26, 28).toInt();
        int parsedSeconds = line.substring(29, 31).toInt();
        Serial.println(String(parsedHours) + ":" + String(parsedMinutes) + ":" + String(parsedSeconds));

        localEpoc = (parsedHours * 60 * 60 + parsedMinutes * 60 + parsedSeconds);
        Serial.println(localEpoc);
        localMillisAtUpdate = millis();
        timeClient.stop();
      }
    }
  }

}
```

wuschel1337, 2022-05-03:
Hello, unfortunately I have the same error, I have now tried 2 Wemos, the stack error comes with both. Is there a fix? Or how do I get the monitor to work?

`*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: (IP unset)
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.38
Signal Strength (RSSI): 100%
Server started
Use this URL : http://192.168.178.38:80/

Unhandled C++ exception: std::bad_alloc
>>>stack>>>

ctx: sys
sp: 3ffff9c0 end: 3fffffb0 offset: 01b0
3ffffb70:  00000000 00000000 00000000 4022d580  
3ffffb80:  40000000 00000000 00000000 40235514  
3ffffb90:  00000000 00000000 00000000 00000000  
3ffffba0:  3fff2240 00000001 3fff9ef4 40235555  
3ffffbb0:  3fff9ef4 40235c54 3fff9ef4 40235ccb  
3ffffbc0:  3fff2240 3ffffe0c 00000220 40235e55  
3ffffbd0:  3fffbecc 3ffffe0c 3fff2240 40222601  
3ffffbe0:  65687314 31796c6c 4638392d 44424134  
3ffffbf0:  39424230 685f0538 04707474 7063745f  
3ffffc00:  636f6c05 00006c61 00000000 00000000  
3ffffc10:  00000000 00000000 00000000 00000000  
3ffffc20:  00000000 00000000 00000000 00000000  
3ffffc30:  00000000 00000000 00000000 00000000  
3ffffc40:  00000000 00000000 00000000 00000000  
3ffffc50:  00000000 00000000 00000000 00000000  
3ffffc60:  00000000 00000000 00000000 00000000  
3ffffc70:  00000000 00000000 00000000 00000000  
3ffffc80:  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 00000000 00000000 00000000  
3ffffca0:  00000000 00000000 00000000 00000000  
3ffffcb0:  00000000 00000000 00000000 00000000  
3ffffcc0:  00000000 00000000 00000000 00000000  
3ffffcd0:  00000000 00000000 00000000 00000000  
3ffffce0:  00210027 3fff8001 00000078 40220022  
3ffffcf0:  3fff2240 3fffbca4 00000001 4021f4ca  
3ffffd00:  00000000 00000000 00000000 00000000  
3ffffd10:  00000000 00000000 00000000 00000000  
3ffffd20:  00000000 00000000 00000000 00000000  
3ffffd30:  00000000 00000000 00000000 00000000  
3ffffd40:  00000000 00000000 00000000 00000000  
3ffffd50:  00000000 00000000 00000000 00000000  
3ffffd60:  00000000 00000000 00000000 00000000  
3ffffd70:  00000000 00000000 00000000 00000000  
3ffffd80:  00000000 00000000 00000000 00000000  
3ffffd90:  00000000 00000000 00000000 00000000  
3ffffda0:  00000000 00000000 00000000 00000000  
3ffffdb0:  00000000 00000000 00000000 00000000  
3ffffdc0:  00000000 00000000 00000000 00000000  
3ffffdd0:  00000000 00000000 00000000 00000000  
3ffffde0:  00000000 00000000 00000000 00000000  
3ffffdf0:  00000000 00000000 00000000 00000000  
3ffffe00:  00000000 00000000 3f000001 00000000  
3ffffe10:  00000005 000014e9 3ffffe60 4022279c  
3ffffe20:  40248400 00000030 00000010 3fff97e4  
3ffffe30:  00000000 3fff2240 3fff2240 40220018  
3ffffe40:  00000000 00000000 00000000 00000000  
3ffffe50:  00000000 00000000 00000000 4024009c  
3ffffe60:  00210000 00050000 00000000 4023abee  
3ffffe70:  00000000 3fff2240 3fff850c 402200b3  
3ffffe80:  3ffec9ec 0033caca 4024871e 3fff3d6c  
3ffffe90:  40000040 00000030 00000010 3fffae14  
3ffffea0:  3fffae46 00000000 3fff850c 402200d4  
3ffffeb0:  00004bc6 40238bb4 00003a98 4022aa47  
3ffffec0:  3fffae46 00000000 3fff8554 40220116  
3ffffed0:  000001ca 40100d7c 3fff8554 4023b260  
3ffffee0:  000014e9 3fff3b0c 3fff3b0c 401012c4  
3ffffef0:  40240000 000001d3 3fffff60 3fff3b0c  
3fffff00:  3fff3bb4 3fffae14 3fffae32 4023f360  
3fffff10:  00000014 00000000 3fff3b0c 40101074  
3fffff20:  3fffdc80 3fff411c 3fff8494 3fff4164  
3fffff30:  00000008 3fff3b0c 3fffae14 40237101  
3fffff40:  3fffdc80 3fff411c 3fff8494 40236f37  
3fffff50:  4025673a 3fff411c 3fff8494 4025674b  
3fffff60:  3fffae24 3fffae14 00000000 3fffdcb0  
3fffff70:  4025140f 00000000 3fff8494 402584d7  
3fffff80:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffff90:  40000e19 40001878 00000001 3fffffb0  
3fffffa0:  3fffff10 aa55aa55 000000b1 40105374  
<<<stack<<<

last failed alloc call: 4022E071(3412)

 ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0fd86a07
~ld

UtcOffset=-7.00
OctoPrintApiKey=
OctoPrintHostName=octopi
OctoPrintServer=
OctoPrintPort=80
OctoAuthUser=
OctoAuthPass=
minutesBetweenDataRefresh=15
themeColor=light-green
IS_BASIC_AUTH=1
www_username=admin
www_password=password
DISPLAYCLOCK=1
IS_24HOUR=0
INVERT_DISPLAY=0
DISPLAYWEATHER=1
WeatherApiKey=
CityID: 5304391
IS_METRIC=0
WeatherLanguage=en
HAS_PSU=0
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: (IP unset)
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.38
Signal Strength (RSSI): 100%
Server started
Use this URL : http://192.168.178.38:80/

Unhandled C++ exception: std::bad_alloc

>>>stack>>>`

Qrome, 2022-05-03:
@wuschel1337 -- looks like yours is crashing when starting up.  Do you have the FS (SPIFF) set for 1M so it has a location to write the configuration?   Also, what version are you running?

wuschel1337, 2022-05-03:
After i using this file, it works very well. 
https://github.com/Qrome/printer-monitor/files/6665707/v3.0_printermonitor.ino.d1_mini_pause_brightness.zip

SohaibSiddique, 2022-05-03:
it was working fine now
its automatically resolved did mot figured out the issue