Qrome/printer-monitor #157: Firmware update page doesnt load

pingywon, 2022-02-07:
Just for the sake of documentation at least, when I click Firmware Update it doesnt load. This is a really cool project and I hope someone picks it up and starts active development again, there a ton of things that could be done with this.

Qrome, 2022-02-07:
Firmware update does work.   You must download the binary file you wish to load or create it first.
[image]

When you see this page then click on the "Choose File" button... select the binary file you want to load, then click on the update button.

closed 2022-02-07 by Qrome

pingywon, 2022-02-07:
> When you see this page then click on the "Choose File" button... select the binary file you want to load, then click on the update button.

Cant ever get to that page, thats what im saying. Tried in a few browsers, incognito and all that.  What am i missing?

Qrome, 2022-02-07:
What do you get when you click on the link for the firmware update?   Is everything else working for you?   Do you know what version of the ESP8266 core you used to compile your build?  What version are you currently running?

pingywon, 2022-02-07:
I just bought (from you id imagine) from Etsy. When I click on the firmware (which to be clear I dont need) the page just refreshes.  Everything else works great! version 3.0