Qrome/printer-monitor #159: Unable to connect to IP address of Octoprint

objectiveSee, 2022-02-17:
Hello, I am able to configure the Settings page to use my Octoprint's IP Address. I can have it run a test which succeeds in pulling API data from the Octoprint. When I return to the main menu though it says that the connection has failed. I have tried restarting the wemos. Any ideas?

**Main Menu:**
[image]

**Config:**
[image]

**API response:**
[image]

Qrome, 2022-02-17:
The test call uses your browsers session and if you have previously authenticated with the octoprint server in your browser, it will show a success.   Do the same test by starting from an incognito window.   Make sure if you are using a username and password with Octoprint that it is correct.

Tosan123ACA, 2022-07-09:
Hola, tengo el mismo problema que el usuario anterior, realizo el procedimiento que expresas y se conecta ok, pero sigue sin realizar conexión la wemos.

anghdezl, 2025-01-11:
> Hola, tengo el mismo problema que el usuario anterior, realizo el procedimiento que expresas y se conecta ok, pero sigue sin realizar conexión la wemos.

Hola buenas, conseguiste solucionarlo? Cuando le doy al boton de Test, me devuelve el JSON con los datos, por lo que la conexion funciona. Incluso Cura con la misma APIKey funciona.  Pero el cacharrin este da error de conexion y no pinta la info en la pantallita.

Tenia la version 2.5 y he actualizado a la 3.0 por si acaso, y nada, dejó de funcionar de la noche a la mañana.

¿Alguna idea?
Saludos y gracias.