Qrome/printer-monitor #161: Screen Not Working | SSD1306 I2C Serial

Brithim, 2022-02-20:
Not sure if I firstly had setup the D1 Board up wrong, Using (https://www.amazon.com/Makerfocus-NodeMcu-Development-ESP8266-Compatible/dp/B01N3P763C/ref=pd_ybh_a_23?_encoding=UTF8&psc=1&refRID=3PNGS61GS9S9AZGEMP7B). The Ocotoprint and web functionality works on it but I've tried many screens and tried to recompile but no luck, I've also ran a scanner script to see and nothing is detected for a screen. Not sure how to fix this

Qrome, 2022-02-20:
Can you share a photo of how you have it wired up and of your board?   Have you confirmed all the correct library files?   The web interface works then?

Brithim, 2022-02-23:
Also adding on to this I've tried to update the D1 with the bin files on the main Git page and still the same outcome

Qrome, 2022-02-23:
I have never run it off of the 3v3 -- always 5V.  I would suspect an issue with the OLED.     Have you been able to check the I2C address of the OLED?  

In the Settings.h file have you tried changing the address for the i2c on the OLED there?

// Display Settings
const int I2C_DISPLAY_ADDRESS = 0x3c; // I2C Address of your Display (usually 0x3c or 0x3d)

Brithim, 2022-02-25:
So I've changed from 3v to 5v, also changed the Adress to 0x3d from c and
no luck I'm thinking there has to be some sort of library issue since I've
tried 2 different screens and two different modules and same outcome but
I'm not sure what I would need to try again or so.

gmsbeak22, 2022-06-25:
I too have the same issue.  WEB interface works perfect, but I have nothing on the display
[image]
y

Qrome, 2022-06-25:
@gmsbeak22 -- please share how you have it wired to the board.   This is not an issue with the software, it is typically incorrect wire / and or pins.

gmsbeak22, 2022-06-25:
Here is how I have the board wired. [image]

Qrome, 2022-06-25:
What OLED board are you using?   Is it the small or larger one?

Qrome, 2022-06-25:
@gmsbeak22 -- looking at the back of your display board, it looks like it shows the I2C address you should use.  Please try selecting that in the Settings.h file:
const int I2C_DISPLAY_ADDRESS = 0x3c; // I2C Address of your Display (usually 0x3c or 0x3d)

Looks like yours might be 0x78 or 0x7a

gmsbeak22, 2022-06-25:
I tried both, no joy.

Qrome, 2022-06-25:
Do you know what drivers it uses?

Did you check this?
//#define DISPLAY_SH1106       // Uncomment this line to use the SH1106 display -- SSD1306 is used by default and is most common

gmsbeak22, 2022-06-25:
I'm going to double-check the drivers.  When I uncomment the DISPLAY_SH1106, I get an error message.  I'm thinking it's missing its specific driver.

gmsbeak22, 2022-06-26:
I tried the address change, tried uncommenting the SH1106. Screen is still blank, but the board functions properly.

Phil2336, 2023-02-15:
Seem to be having a very similar issue here.

Have a 1.3 inch Display, SSD1106. Test against an ESP32 & known to be working.

It has address setting & have changed the code to reflect the same.

`// const int I2C_DISPLAY_ADDRESS = 0x3c; // I2C Address of your Display (usually 0x3c or 0x3d)`
`const int I2C_DISPLAY_ADDRESS = 0x78;`
`const int SDA_PIN = D2;`
`const int SCL_PIN = D5; // original code D5 -- Monitor Easy Board use D1`
`boolean INVERT_DISPLAY = false; // true = pins at top | false = pins at the bottom`
`#define DISPLAY_SH1106       // Uncomment this line to use the SH1106 display --`

Web Interface is working fine.
Wiring triple checked; Running on the 5V pin.

Image of the Display.

[image]

Qrome, 2023-02-15:
@Phil2336 -- did you try removing the comment on this line like the file says?

`#define DISPLAY_SH1106       // Uncomment this line to use the SH1106 display`

Remove the # from that line as the 1.3 inch display needs that line.

Phil2336, 2023-02-15:
Do also have a 1306 here & tried that also with the commented line back in, but still no go.

Interesting Both size displays I have have that 0x78/0x7J jumper & also the Wemos has a FTI chip it seems instead of a CH340.

Removing the # causes
`Compilation error: 'define' does not name a type`

Qrome, 2023-02-15:
Make sure you have version 4.1.0 as mentioned in the readme file.   Since your web interface is working try updating to the recompiled binary for the 1.3 display.

"SSD1306Wire.h" --> https://github.com/ThingPulse/esp8266-oled-ssd1306/releases/tag/4.1.0 (version 4.1.0)

Phil2336, 2023-02-15:
Now Working Fine!
But What a Rabbit Hole....

Changed the Binary  to 4.1.0 but that didn't fix it.

Loaded some I2C code on the chip & ran that & it reported the display as found at 0x3c.
Of course I changed the code as my display was labeled 0x78.

Comparing the 2 addresses.....
`0x3c  0011 1100`
`0x78 0111 1000`

Shifted One Bit.

https://community.element14.com/members-area/personalblogs/b/blog/posts/oled-i2c-silkscreens-are-wrong-or-are-they

Thanks for your help.