Qrome/printer-monitor #163: Hi got a issue

KSTV28, 2022-09-25:
So I been fighting with this project for a few day's now where I said heck with it, I bought the Wemos Mini D2 and the HiLetgo 1.3" IIC I2C Serial 128x64 SSH1106 SSD1306 OLED LCD and when I go to the settings.h and do must uncomment #define DISPLAY_SH1106 in the Settings.h it still does the lines on the screen, All my wires on the Wemos Mini is correct when I go in and change the wording it turns the screen completely off and when I revert back it come's on with the lines or sometimes it has like tv static so I have no idea what is going on hopefully someone could help me figure this out cause I would like to get this to work.  
[image]

Qrome, 2022-09-25:
Are you sure you have the 1.3" version?  Is the web interface working?   Do the settings save?

KSTV28, 2022-09-25:
I can post the amazon link to the screen I ordered that you had posted for the 1.3

Qrome, 2022-09-25:
The case you are showing there appears to be for the smaller display so you would comment out the SSH1106 driver on line 97 of the Settings.h file.

KSTV28, 2022-09-25:
I went with the bigger screen the 1.3 and found a remix 3D model of the case for the bigger screen https://www.thingiverse.com/thing:2884823

Qrome, 2022-09-25:
Is the Web interface working?  Does it save and store settings?

KSTV28, 2022-09-25:
The web interface is working

KSTV28, 2022-09-25:
I see stuff is moving on the screen like time and what not but it still has those white lines moving

Qrome, 2022-09-25:
Have you tried loading the pre-compiled binary for the 1106 display?   https://github.com/Qrome/printer-monitor/blob/master/printermonitor.ino.d1_mini_SH1106.bin

KSTV28, 2022-09-25:
https://user-images.githubusercontent.com/73322561/192124826-c8171eba-0218-478e-89cf-17832dfa5a24.mov

KSTV28, 2022-09-25:
I have not tried that yet
how do I compile that

Qrome, 2022-09-25:
you can load it through the web interface

KSTV28, 2022-09-25:
under firmware update?

KSTV28, 2022-09-25:
also what is the main login and password or do I have to setup a custom one?

Qrome, 2022-09-25:
user: admin 
password:   password

This is the same as the web interface

KSTV28, 2022-09-25:
when I try to login to the web interface its brings up an error must to be something on my end

Qrome, 2022-09-25:
have you been able to change and save any settings at all ever?   
If not, then make sure you compile it with allocation for the file system to create the configuration settings on it.
[image]

KSTV28, 2022-09-25:
so I need the flash size to be 4MB (FS:2MB and not the 1MB

KSTV28, 2022-09-25:
same thing This site can’t be reached The connection was reset ugh this getting on my nerves

KSTV28, 2022-09-25:
Dude thank you so much after I updated the firmware to the printermonitor.ino.d1_mini_SH1106.bin it works nomrally thank you very much

KSTV28, 2022-09-25:
now a other question how can I change the time from UTC to CST

Qrome, 2022-09-25:
Looks like you may not have had the right library loaded for the display when you compiled it.

closed 2022-09-25 by Qrome