Qrome/printer-monitor #164: Blank/Black Screen on SH1106

m0uh85, 2022-10-30:
If I use the pre-compiled file of SH1106 its working but if I compile it by myself the screen keeps blank/black (Server is running, Web-interface and so on also....) so wireing is correct.

SH1106 is right defined... Display adress i tried both... any idea? Because I would like to use the moonraker (https://github.com/truckershitch/printer-monitor) but there is the same issue... the pre compiled file for moonraker is also not working.

Qrome, 2022-10-30:
My guess is you don't have a compatible SH1106 library locally.  Perhaps you may need to verify the library files you have installed locally.   The compiled version has it.

m0uh85, 2022-10-30:
> My guess is you don't have a compatible SH1106 library locally. Perhaps you may need to verify the library files you have installed locally. The compiled version has it.

Can you tell me which library you have used?

Do you have the possibility to test the files for the "moonraker" Version? Just if the Display is working?

m0uh85, 2022-11-18:
Any idea?

Qrome, 2022-11-18:
You need to make sure you have the full library that will support the display you used.   The binary has it because it was compiled with that library.

This is NOT a code issue.  This is a library / configuration issue.

m0uh85, 2022-11-19:
> You need to make sure you have the full library that will support the display you used. The binary has it because it was compiled with that library.
> 
> This is NOT a code issue. This is a library / configuration issue.

Can you tell me which library u have used to compile it?

Qrome, 2022-11-19:
From what I can tell it is just the one listed in the readme.md file:
https://github.com/ThingPulse/esp8266-oled-ssd1306/releases/tag/4.1.0 (version 4.1.0)

m0uh85, 2022-11-20:
> From what I can tell it is just the one listed in the readme.md file: https://github.com/ThingPulse/esp8266-oled-ssd1306/releases/tag/4.1.0 (version 4.1.0)
> 
> This library also includes the SH1106

Thanks. I had v4.2 & 4.3 tested of the same library - not working
With 4.1 everything is working fine!

closed 2022-11-20 by m0uh85