Qrome/printer-monitor #166: Has this project been abandoned?

ArcAIN6, 2023-02-16:
it's been several years since anything has been update within the project, and since the last update (3 years ago) many things have changed. it is no longer possible to build this, without completely corrupting an updated IDE environment with loads of outdated libraries, board configs, and settings.

closed 2023-02-16 by Qrome

ArcAIN6, 2023-02-16:
i'll take that as a yes, it's been abandoned

Qrome, 2023-02-16:
@ArcAIN6 -- not sure what you are looking for.  Sounds like you had an issue with the code.  If you want help, post it.  As for your comments, it compiles with all the latest libraries except the 1306 display.   Also, you don't need to use the settings.h as all the settings are stored using the web interface.   The project works.   Do you need help?   Other than asking if it is dead?  If you looked at some of the posts, there are people still building it.

Qrome, 2023-02-16:
I closed the issue created hear because it was simply a statement of your frustration.

ArcAIN6, 2023-02-17:
the code base is completely out of date, the libraries linked against it have made significant enough changes that they won't work with the code, as well as there are several issues with the code itself. It would require a ton of work to bring it up to the current standards of the IDE, and the libraries used. Since there's been absolutely NO movement on the code in 3 years, one can assume the code has been abandoned. The main subject of the OP asked if the code was abandoned, and the rest of the post explained why the question was asked. but that's fine. Good day