Qrome/printer-monitor #167: Display never turns on

MeeMKade, 2023-03-27:
I have the board compiled with the code and I'm able to set it up, but the display never turns on. Ive made sure to use the right versions, compiled it multiple times, resoldered and unsoldered the connections, but nothing ever works. What do I do?

MeeMKade, 2023-03-29:
It didnt seem to work, what should i do now?

morphias2004, 2023-04-23:
> It didnt seem to work, what should i do now?

What screen are you using?

If you have the 1.3", you need to uncomment a line in the settings.h as per the install guide.

MeeMKade, 2023-04-24:
This:
https://www.amazon.com/dp/B00O2LLT30?ref=ppx_yo2ov_dt_b_product_details&th=1

MeeMKade, 2023-04-26:
I also moved #define DISPLAY_SH1106 to the top, but its in orange for some sort of reason. I moved it to the top because I think it was giving me some sort of error if i left it where it originally was. 
[image]